Arm-Examples/ML-examples #3: '-larmnn' and '-larmnnTfParser' not found

rainbowkenny, 2018-06-18:
Hi, I followed your steps in the ML-examples/armnn-mnist/
After I executed 'make test', I got the following error messages:

/usr/bin/x86_64-linux-gnu-ld: cannot find -larmnn
/usr/bin/x86_64-linux-gnu-ld: cannot find -larmnnTfParser

Can you please suggest how to fix it?

curtywang, 2018-07-11:
you need to change the ARMNN_LIB and ARMNN_INC lines in the Makefile to point to your libraries.

phamyphen, 2018-07-12:
What is the expected lib for ARMNN_LIB and ARMNN_INC directory? 

There is no build-x86_64/release/armnn folder under armnn-devenv as README file illustrated.

BTW, I am building it with Android NDK.

curtywang, 2018-07-12:
ARMNN_LIB is going to be your /<path-to-armnn-sources>/armnn/build/ folder as you do in the example here: https://developer.arm.com/technologies/machine-learning-on-arm/developer-material/how-to-guides/configuring-the-arm-nn-sdk-build-environment-for-tensorflow/single-page

and ARMNN_INC is going to be /<path-to-armnn-sources>/armnn/include/

phamyphen, 2018-07-12:
Thanks curtywang.

I added the lib as you mentioned. However, below errors popped up.

                 from /home/meng1.yu/armnn-devenv/armnn/include/armnn/ArmNN.hpp:9,
                 from mnist_tf.cpp:13:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr const char* armnn::GetStatusAsCString(armnn::Status)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:25:1: error: body of constexpr function ‘constexpr const char* armnn::GetStatusAsCString(armnn::Status)’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr const char* armnn::GetComputeDeviceAsCString(armnn::Compute)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:36:1: error: body of constexpr function ‘constexpr const char* armnn::GetComputeDeviceAsCString(armnn::Compute)’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr const char* armnn::GetActivationFunctionAsCString(armnn::ActivationFunction)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:54:1: error: body of constexpr function ‘constexpr const char* armnn::GetActivationFunctionAsCString(armnn::ActivationFunction)’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr const char* armnn::GetPoolingAlgorithmAsCString(armnn::PoolingAlgorithm)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:65:1: error: body of constexpr function ‘constexpr const char* armnn::GetPoolingAlgorithmAsCString(armnn::PoolingAlgorithm)’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr const char* armnn::GetOutputShapeRoundingAsCString(armnn::OutputShapeRounding)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:75:1: error: body of constexpr function ‘constexpr const char* armnn::GetOutputShapeRoundingAsCString(armnn::OutputShapeRounding)’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr const char* armnn::GetPaddingMethodAsCString(armnn::PaddingMethod)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:86:1: error: body of constexpr function ‘constexpr const char* armnn::GetPaddingMethodAsCString(armnn::PaddingMethod)’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr unsigned int armnn::GetDataTypeSize(armnn::DataType)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:97:1: error: body of constexpr function ‘constexpr unsigned int armnn::GetDataTypeSize(armnn::DataType)’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In instantiation of ‘constexpr bool armnn::StrEqual(const char*, const char (&)[N]) [with int N = 7]’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:112:31:   required from here
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:108:1: error: body of constexpr function ‘constexpr bool armnn::StrEqual(const char*, const char (&)[N]) [with int N = 7]’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr armnn::Compute armnn::ParseComputeDevice(const char*)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:128:1: error: body of constexpr function ‘constexpr armnn::Compute armnn::ParseComputeDevice(const char*)’ not a return-statement
 }
 ^
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp: In function ‘constexpr const char* armnn::GetDataTypeName(armnn::DataType)’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/TypesUtils.hpp:139:1: error: body of constexpr function ‘constexpr const char* armnn::GetDataTypeName(armnn::DataType)’ not a return-statement
 }
 ^
In file included from /usr/include/c++/4.8/bits/char_traits.h:39:0,
                 from /usr/include/c++/4.8/ios:40,
                 from /usr/include/c++/4.8/ostream:38,
                 from /usr/include/c++/4.8/iostream:39,
                 from mnist_tf.cpp:6:
/usr/include/c++/4.8/bits/stl_algobase.h: In instantiation of ‘bool std::equal(_IIter1, _IIter1, _IIter2, _BinaryPredicate) [with _IIter1 = const unsigned int*; _IIter2 = const unsigned int*; _BinaryPredicate = const unsigned int*]’:
/home/meng1.yu/armnn-devenv/armnn/include/armnn/Types.hpp:136:69:   required from here
/usr/include/c++/4.8/bits/stl_algobase.h:1062:46: error: ‘__binary_pred’ cannot be used as a function
  if (!bool(__binary_pred(*__first1, *__first2)))
                                              ^
make: *** [mnist_tf] Error 1

phamyphen, 2018-07-13:
Thanks curtywang

The above compilation issues was solved by changing the compiler to clang++ and setting related libs in the Makefile.

closed 2018-07-16 by mark-arm

hashtagrk, 2020-05-13:
Which Makefile to edit? There are so many of them