Arm-Examples/ML-examples #7: Wrong col_buffer size/type ?

iabdalkader, 2018-10-03:
Hi, the attached network causes memory corruption when run a few times on Cortex-M7. I think the col_buffer size/type is wrong. The generated code allocates a col_buffer of 800, but gdb break on writes almost twice the size, doubling the col_buffer size fixes the issue for now.

[square.pkl.zip](https://github.com/ARM-software/ML-examples/files/2442933/square.pkl.zip)

navsuda, 2018-10-03:
@iabdalkader,
Please look at the output of these [print](https://github.com/ARM-software/ML-examples/blob/master/cmsisnn-cifar10/code_gen.py#L190) statements to understand which NN layer is causing this issue and report it. Can you also share your square_train_test.prototxt and give details on which layer gdb shows the error?

iabdalkader, 2018-10-10:
@navsuda Hi, were you able to reproduce the bug ?

navsuda, 2018-10-10:
@iabdalkader,
Not really (need caffemodel too to reproduce it), but your debug log was sufficient to find the bug. Pushed a fix, see if it helps. 
It should generate
`Layer: ip1, required memory: 1024, im2col buffer size: 1024`

iabdalkader, 2018-10-10:
@navsuda Thanks! I also [fixed](https://github.com/openmv/openmv/commit/4449cbe36968a2a405b9f7353041d1ad27dfd0c5) my C code the same way, and I can confirm it fixes the issue and my test networks run fine.

closed 2018-10-10 by iabdalkader