Arm-Examples/ML-examples #8: Running TfLite model using armnn

nishthagoel12, 2018-10-29:
Hi,
How can I run my TfLite model using armnn?

I am using Quantized MobileNet model used by armnn to test TfLiteParser but my output is something like 

0
2.57139e-39
0
0
0
0
9.18355e-41
0
0
0
0
0
0
0
0
2.35099e-38
0
2.35099e-38
0
0
0
2.40609e-38
3.67342e-40
0

SaoirseARM, 2018-11-08:
Hi @nishthagoel12 , thanks so much for your interest. I wonder could you provide some more information on the issue above i.e. how you have tried the above and any other log information you have to hand. It would be much appreciated to get to the bottom of this issue.

MatthewARM, 2018-11-22:
Hi @nishthagoel12, just to check, can you use Tensorflow Lite's "label_image" sample to verify that the model and the test data are OK?

mark-arm, 2019-01-24:
Closing due to inactivity.

closed 2019-01-24 by mark-arm

damiangajewski, 2019-03-28:
Any solution for that?
I have very similar problem. I have quantized model trained in tensorflow on mnist dataset. When i run it in armnn almost all predictions are 0. This is my code:

``` python
//
// Copyright © 2018 Arm Ltd. All rights reserved.
// See LICENSE file in the project root for full license information.
//

#include <iostream>
#include <string>
#include <fstream>
#include <vector>
#include <memory>
#include <array>
#include <algorithm>
#include <cstring>
#include "armnn/ArmNN.hpp"
#include "armnn/Exceptions.hpp"
#include "armnn/Tensor.hpp"
#include "armnn/INetwork.hpp"
#include "armnnTfLiteParser/ITfLiteParser.hpp"

#include "mnist_loader.hpp"

#define MAX_IMAGES 3000 

// Helper function to make input tensors
armnn::InputTensors MakeInputTensors(const std::pair<armnn::LayerBindingId,
    armnn::TensorInfo>& input,
    const void* inputTensorData)
{
    return { { input.first, armnn::ConstTensor(input.second, inputTensorData) } };
}

// Helper function to make output tensors
armnn::OutputTensors MakeOutputTensors(const std::pair<armnn::LayerBindingId,
    armnn::TensorInfo>& output,
    void* outputTensorData)
{
    return { { output.first, armnn::Tensor(output.second, outputTensorData) } };
}

int main(int argc, char** argv)
{
    // Default input size is one image
    unsigned int nrOfImages = 1;

    // Optimisation mode 0 = CPU Reference (unoptimised)
    // Optimisation mode 1 = CPU Accelerator
    // Optimisation mode 2 = GPU Accelerator
    unsigned int optimisationMode = 1;

    // Check program arguments
    if (argc != 3)
    {
      std::cout << "Invalid arguments. Exactly 2 arguments required." << std::endl;
      std::cout << "Example: ./mnist_tf 1 10" << std::endl;
      std::cout << "Optimisation modes: 0 for CpuRef, 1 for CpuAcc, 2 for GpuAcc" << std::endl;
      std::cout << "Input size: 1 to 2000 (number of images to predict)" << std::endl;
      return 1;
    }

// Parse input size option
    nrOfImages = std::stoi(argv[2]);
    if (!(nrOfImages > 0 && nrOfImages <= MAX_IMAGES))
    {
      std::cout << "Error: Maximum number of images is " << MAX_IMAGES << std::endl;
      return 1;
    }

    // Parse optimisation mode option
    optimisationMode = std::stoi(argv[1]);
    if (!(optimisationMode == 0 || optimisationMode == 1 || optimisationMode == 2))
    {
      std::cout << "Invalid optimisation mode." << std::endl;
      return 1;
    }

    // Import the TensorFlow model. Note: use CreateNetworkFromBinaryFile for .pb files.
    armnnTfLiteParser::ITfLiteParserPtr parser = armnnTfLiteParser::ITfLiteParser::Create();
    armnn::INetworkPtr network = parser->CreateNetworkFromBinaryFile("model/quantized_model_dense.tflite");

    // Find the binding points for the input and output nodes
    armnnTfLiteParser::BindingPointInfo inputBindingInfo = parser->GetNetworkInputBindingInfo(0, "dense_1_input");
    armnnTfLiteParser::BindingPointInfo outputBindingInfo = parser->GetNetworkOutputBindingInfo(0, "activation_3/Softmax");

    // Create ArmNN runtime
    armnn::IRuntime::CreationOptions options; // default options
    armnn::IRuntimePtr runtime = armnn::IRuntime::Create(options);

    // Optimize the network for a specific runtime compute device, e.g. CpuAcc, GpuAcc
    std::cout << "Optimisation mode: ";
    armnn::Compute device;
    switch(optimisationMode)
    {
      case 0: device = armnn::Compute::CpuRef; std::cout << "CpuRef" << std::endl; break;
      case 1: device = armnn::Compute::CpuAcc; std::cout << "CpuAcc" << std::endl; break;
      case 2: device = armnn::Compute::GpuAcc; std::cout << "GpuAcc" << std::endl; break;
    }

    armnn::IOptimizedNetworkPtr optNet = Optimize(*network, {device}, runtime->GetDeviceSpec());

    // Load the optimized network onto the runtime device
    armnn::NetworkId networkIdentifier;
    runtime->LoadNetwork(networkIdentifier, std::move(optNet));

    // Load multiple images from the data directory
    std::string dataDir = "data/";

    auto input = new float[nrOfImages][g_kMnistImageByteSize];
    auto output = new float[nrOfImages][10];
    auto labels = new int[nrOfImages];

for (int i = 0; i < nrOfImages; ++i)
    {
      std::unique_ptr<MnistImage> imgInfo = loadMnistImage(dataDir, i);
      if (imgInfo == nullptr)
          return 1;
      for (int j = 0; j < 784; ++j) input[i][j] -= 33;
      std::memcpy(input[i], imgInfo->image, sizeof(imgInfo->image));
      labels[i] = imgInfo->label;
    }

  for (int i = 0; i < nrOfImages; ++i)
    {
        for (int j = 0; j < 784; ++j) {
            std::cout << input[i][j] << " ";
        }
        std::cout << std::endl << std::endl;
    }

    // Execute network
    armnn::InputTensors inputTensor = MakeInputTensors(inputBindingInfo, &input[0]);
    armnn::OutputTensors outputTensor = MakeOutputTensors(outputBindingInfo, &output[0]);

    armnn::Status ret = runtime->EnqueueWorkload(networkIdentifier, inputTensor, outputTensor);

    // Check output and compute correct predictions
    int nrOfCorrectPredictions = 0;
    for (int i = 100; i < nrOfImages; ++i)
    {
      float max = output[i][0];
      int label = 0;

      for (int j = 0; j < 10; ++j)
      {
        // Translate 1-hot output to find integer label
        if (output[i][j] > max)
        {
          max = output[i][j];
          // std::cout << max << std::endl;
          label = j;
        }
      }
      if (label == labels[i]) nrOfCorrectPredictions++;
      std::cout << "#" << i+1 << " | Predicted: " << label << " Actual: " << labels[i] << std::endl;
    }
    std::cout << "Prediction accuracy: " << (float)nrOfCorrectPredictions/nrOfImages*100 << "%";
    std::cout << std::endl;

    delete[] input;
    delete[] output;
    delete[] labels;

    return 0;
}
```
Is it something wrong with that code?

MatthewARM, 2019-03-29:
Hi @damiangajewski I think maybe the input tensor isn't quantized according to what the model expects...

Can you call runtime->GetInputTensorInfo(networkIdentifier, inputBindingInfo) and then call GetQuantizationScale() and GetQuantizationOffset() on the TensorInfo that you get? That will tell you how the model will interpret your data.  Alternatively you can use TensorBoard or whatever to examine the model file.

Then when you prepare your input tensor, you'll need to convert the bitmap data from the image file into floating point, then into QAsymm8 according to those quantization parameters.

Hope that helps,
Matthew

damiangajewski, 2019-03-29:
Thanks for reply @MatthewARM.
My input scale is 0.012727910652756691 and offset is 33. As I have images in ranges 0-255, is it necessary to convert data?

So to summary, I have to create input tensor of type QAsymm8 (with quantization parameters) and then convert data to uint_8, yes?