Arm-Examples/ML-examples #9: compile error of minst_tf for arm32 device

yanghaiping1993, 2018-11-15:
I am trying to compile minst_tf for arm32 device, and Makefile add the 
ARMNN_LIB=${HOME}/armnn-devenv/armnn/build
ARMNN_INC=${HOME}/armnn-devenv/armnn/include

and the compile command is:
g++ -O3 -m32 sstd=c++14 -I$(ARMNN_INC) mnist_tf.cpp -o minist_tf -L$(ARMNN_LIB) -larmnn -larmnnTfParse -lpthread
but i got the error:
/usr/bin/ld: skipping incompatible /home/happyyang/armnn-devenv/armnn/build/libarmnn.so when searching for -larmnn
/usr/bin/ld: cannot find -larmnn
/usr/bin/ld: skipping incompatible /home/happyyang/armnn-devenv/armnn/build/libarmnnTfParser.so when searching for -larmnnTfParser
/usr/bin/ld: cannot find -larmnnTfParser
collect2: error: ld returned 1 exit status
make: *** [mnist_tf] Error 1

yanghaiping1993, 2018-11-15:
@SaoirseARM

SaoirseARM, 2018-11-15:
Hi @yanghaiping1993 , thanks for getting in touch.
Were those library files i.e. libarmnn.so/libarmnnTfParser.so compiled in 64 bit by any chance?  That could explain the incompatibility errors you are seeing above.

yanghaiping1993, 2018-11-19:
@SaoirseARM it is 32 bit lib, and the error is made by choosing wrong compiler，and I choose the arm-linux-androideabi-clang/clang++，reedit the Makefile，it works on arm32 device (android OS) if I compile with cmake command:-DARMCOMPUTENEON=0 -DARMCOMPUTECL=0. BUT,  if I compile with cmake command:-DARMCOMPUTENEON=1 -DARMCOMPUTECL=0, the log are showed as:
Running 542 test cases...
unknow location(0): fatal error: in "NeonTimerInstrument/NeonTimerMeasure": std::bad_alloc: std::bad_alloc
/armnn-devenv/armnn/src/armnn/test/NeonTimerTest.cpp(35): last checkpoint: "NeonTimerMeasure" entry

*** 1 failuer iis detected in the test module "UnitTests"
My android device is an armv7a 32 bit cpu, and RAM is 2GB, what can cause that error?

huanzoey, 2019-07-09:
I am trying to compile minst_and the Makefile is
ARMNN_LIB = /home/zoey/zoey/armnn/build
ARMNN_INC = /home/zoey/zoey/armnn/include
all: mnist_tf
mnist_tf: mnist_tf.cpp mnist_loader.hpp
	g++ -std=c++14 -I$(ARMNN_INC) mnist_tf.cpp -o mnist_tf -L$(ARMNN_LIB) -larmnn -larmnnTfParser -lpthread
clean:
	-rm -f mnist_tf 
test:  mnist_tf
	LD_LIBRARY_PATH=${LD_LIBRARY_PATH}:$(ARMNN_LIB) ./mnist_tf
it happened:
g++ -std=c++14 -I/home/zoey/zoey/armnn/include mnist_tf.cpp -o mnist_tf -L/home/zoey/zoey/armnn/build -larmnn -larmnnTfParser -lpthread
/usr/bin/ld: 当搜索用于 /home/zoey/zoey/armnn/build/libarmnn.so 时跳过不兼容的 -larmnn
/usr/bin/ld: 找不到 -larmnn
/usr/bin/ld: 当搜索用于 /home/zoey/zoey/armnn/build/libarmnnTfParser.so 时跳过不兼容的 -larmnnTfParser
/usr/bin/ld: 找不到 -larmnnTfParser
collect2: error: ld returned 1 exit status
Makefile:7: recipe for target 'mnist_tf' failed
make: *** [mnist_tf] Error 1
my compiler is aarch64-linux-gnu-g++,my device is 64bit,can you help me what cause that error.