Arm-Examples/ML-examples #11: How can I input this argument "const std::map<std::string, armnn::TensorShape>& inputShapes"

helloyongyang, 2019-02-26:
`    virtual armnn::INetworkPtr CreateNetworkFromBinaryFile(
        const char* graphFile,
        const std::map<std::string, armnn::TensorShape>& inputShapes,
        const std::vector<std::string>& requestedOutputs) = 0;`

How can I input this argument "const std::map<std::string, armnn::TensorShape>& inputShapes" in CreateNetworkFromBinaryFile?
I found that inputshape can be {} which is empty in the mnist sample
`    armnnCaffeParser::ICaffeParserPtr parser = armnnCaffeParser::ICaffeParser::Create();
    armnn::INetworkPtr network = parser->CreateNetworkFromBinaryFile("model/lenet_iter_9000.caffemodel",
                                                                   { }, // input taken from file if empty
                                                                   { "prob" }); // output node`

I want to know what this argument is used for, and how should I use it. Could you give me some examples? Thank you!!!

MatthewARM, 2019-02-26:
Hi @helloyongyang, I think it's safe to leave it at {}, in which case the ArmNN network will take its input size from the Caffe file.

If you want to override the value in the Caffe file, then you should provide a map from layer top to `TensorShape`. E.g. if you have one input in your caffe model called `data`, you can do:

std::map<std::string, armnn::TensorShape> shapeOverrides;
shapeOverrides["data"] = armnn::TensorShape(1,3,255,255);
armnn:INetworkPtr newNet = parser.CreateNetworkFromBinaryFile(path, shapeOverrides, {"output"});

This will force the parser to treat `data` as a Tensor of shape {1,3,255,255}.

Our intention is that this argument will be used in the future to set the size of networks at load time, for example by overriding the batch size, or setting the shape of a fully-convolutional network. In the current release it will just perform error checking before any buffers are created - this might be useful in the case where a network is loaded from disk in a secure application.

I hope this answers the question,
Matthew

MatthewARM, 2019-04-26:
Hi @helloyongyang I hope that all made sense, good luck with everything.

closed 2019-08-05 by mark-arm