Arm-Examples/ML-examples #13: ImportError: no module named tensorflow

JobinC, 2019-03-02:
I am following the tutorial on yeah-world and have imported tensorflow and verified it was installed. 
when running 

`python train.py example/model.h5 example/yeah example/sitting example/random`

I received a import error: no module named tensorflow

corrected this by : 
`python3 train.py example/model.h5 example/yeah example/sitting example/random`

now receiving 
`UnicodeDecodeError: 'ascii' codec can't decode byte 0x8d in position 1:ordinal not in range(128)`

sunnypinto, 2019-07-19:
Getting same error. Tried replacing cPickle with pickle, however the load() function stopped working. Pls help

mark-arm, 2020-04-22:
Try "pip install tensorflow" instead of "pip3 install tensorflow"