Arm-Examples/ML-examples #14: Cifar10 Check failed: mdb_status == 0 (2 vs. 0) No such file or directory

marcin-ch, 2019-04-19:
Hello All,

section [Troubleshooting](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/troubleshooting) shows solution for error:
**Check failed: mdb_status == 0 (2 vs. 0) No such file or directory**
The solution is:

`- - - - - -`
Open file
`gedit ~/CMSISNN_Webinar/ML-examples/cmsisnn-cifar10/models/cifar10_m7_train_test.prototxt`

and check that the ***mean_file*** and ***source*** are pointing to the files in *caffe/examples/cifar10/* directory as shown below
```
# The following lines will have to be updated.
# Note: You will have to replace  with the path where you have cloned the Caffe repository

mean_file: "/caffe/examples/cifar10/mean.binaryproto"
source: "/caffe/examples/cifar10/cifar10_train_lmdb"
source: "/caffe/examples/cifar10/cifar10_test_lmdb"
```
`- - - - - -`

However, files mentioned above ie
***mean.binaryproto***,
***cifar10_train_lmdb***,
***cifar10_test_lmdb***
do not exist in *caffe/examples/cifar10/*

so how can I get them? Should have they been in *caffe/examples/cifar10/* already?

Looking at *Caffe* [GitHub](https://github.com/BVLC/caffe/tree/master/examples/cifar10) repo, I do not see these files.

Hints more than welcome.

navsuda, 2019-04-24:
@marcin-ch, please follow the steps in "Prepare the Dataset" in the [Caffe Cifar10-tutorial](http://caffe.berkeleyvision.org/gathered/examples/cifar10.html)

marcin-ch, 2019-04-29:
@navsuda thank you very much for support, indeed this is what I was looking for!

closed 2019-04-29 by marcin-ch