Arm-Examples/ML-examples #16: CMSIS-NN CIFAR10 example for STM32F746G-DISCO does not work?

marcin-ch, 2019-04-29:
Hello All,

I am following this [Image recognition on Arm Cortex-M with CMSIS-NN guide](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/before-you-begin)

I have exactly the same hardware:
```
STM32F746G-DISCO
STM32F4DIS-CAM
```

the same software installed
```
Ubuntu 16.04 LTS
Python 2.7.12
Caffe
GNU Tools for Arm Embedded Processors 7-2017-q4-major
```

and I am able to reproduce all mentioned in the guide steps (including building basic camera app, quantizing the model, converting the model), **except the final one! = Deploy transformed model on an Arm Cortex-M processor**

The final call is:
```
#Run this command in cmsisnn_demo folder
mbed compile -m DISCO_F746NG -t GCC_ARM --source . --source ../ML-examples/cmsisnn-cifar10/code/m7 --source ../ML-examples/cmsisnn-cifar10/camera_demo/camera_with_nn/ --source ../CMSIS_5/CMSIS/NN/Include --source ../CMSIS_5/CMSIS/NN/Source --source ../CMSIS_5/CMSIS/DSP/Include --source ../CMSIS_5/CMSIS/DSP/Source --source ../CMSIS_5/CMSIS/Core/Include -j8
```
The only difference is in 
```
--source ../ML-examples/cmsisnn-cifar10/camera_demo/camera_with_nn/
```
the original call refers to
```
--source ../ML-examples/cmsisnn-cifar10/camera_demo/camera_app/
```

But I have not changed *camera_app* folder, just used already prepared by ARM folder with *camera_with_nn* (ARM provides ready-to-go camera apps without NN and with NN, [here](https://github.com/ARM-software/ML-examples/tree/master/cmsisnn-cifar10/camera_demo))

And mbed finishes with info
```
[mbed] ERROR: "/usr/bin/python" returned error.
```
(I know, not much descriptive... but I do not see any ERRORs during building, only: `multiple definition of` or `first defined here`)

So perhaps any known issues reffering to above?
Has anyone tried to complete this guide?

Hints more than welcome.

marcin-ch, 2019-05-02:
Not sure if I call it a workaround but finally I have this example running.
I just changed the final call to:
```
#Run this command in cmsisnn_demo folder
mbed compile -m DISCO_F746NG -t GCC_ARM --source . --source ../ML-examples/cmsisnn-cifar10/code/m7 --source ../ML-examples/cmsisnn-cifar10/camera_demo/camera_with_nn/ --source ../CMSIS_5/CMSIS/NN/Include --source ../CMSIS_5/CMSIS/NN/Source --source ../CMSIS_5/CMSIS/DSP/Include --source ../CMSIS_5/CMSIS/Core/Include -j8
```

it means I just simply cut this one:
```
--source ../CMSIS_5/CMSIS/DSP/Source
```

marcin-ch, 2019-05-22:
just to close the issue and prove that I went through all the steps and got working demo :satisfied:
[image]

closed 2019-05-22 by marcin-ch

youngolutosin, 2019-07-06:
Hello marcin_ch, I am currently working on this project for my thesis. Please I need help getting it through. Do you have the link to step by step procedures so that I can follow?

marcin-ch, 2019-07-10:
Hi @youngolutosin ,
I followed this guide [Image recognition on Arm Cortex-M with CMSIS-NN guide](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/before-you-begin) as mentioned in my first post here.

youngolutosin, 2019-07-11:
I am going through the steps and I was able to Implement the first basic camera example. I am stucked in check failed:mdb_status == 0 (2 vs.0) No file or directory. I have edited the mean_file and source file but in caffe/exmaples/cifar10 directory, I don't have mean.binaryproto, cifar10_train_lmdb and cifar10_test_lmdb. Please how did you solve the problem.

marcin-ch, 2019-07-15:
Hi @youngolutosin ,
first of all, make sure you double-checked [Troubleshooting guide](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/troubleshooting)

And then have a look at issue I created [here](https://github.com/ARM-software/ML-examples/issues/14) with the same question as you. Hope that helps.

youngolutosin, 2019-07-15:
Thank you for your response. I already figured out the problem.  My project
is working now. Seems the model is sensitive to some pictures. I an working
on how to train the model with my own dataset and reduce the class to the
specific I wanted.

Thank you and I really appreciate.

Regards,
Young.

marcin-ch, 2019-07-15:
@youngolutosin ,
yup, model is not very reliable, but they mentioned about it in the guide:
>Note: that this model is very easily affected by changes in light conditions, we have chosen it only for simplicity.

And what may be also helpful for you is the keyword: `transfer learning`.
In simple words you can use some already existing/trained model for image classification and re-train it using your dataset. It should greatly shorten the amount of time needed to train the model.

Hope that helps and good luck!

shell0108, 2020-04-06:
@marcin-ch @youngolutosin Hello ! I'm doing the same CMSIS_NN CIFAR10 example ,following [the same guide](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/set-up-environment)   ,but I met some problems.

It seems that you have done it successfully, and I will be very grateful if you offered some help!
here is [my _issue](https://github.com/ARM-software/ML-examples/issues/37),
[image]

joshbindels, 2020-11-08:
Hi @youngolutosin @marcin-ch @shell0108 ,

I'm currently working on this project for my thesis as well.
 
Did you notice a difference when running the model on the board or on a laptop?
When I run accuracy tests using caffe or as a python script on my host machine I'm getting a decent accuracy. But, when I run a similar accuracy test on the board the accuracy is quite low, ~30%.

youngolutosin, 2020-11-08:
Hi,

The accuracy of the cifar model after optimization will obviously reduce
due to the conversion of the model from float32 into fixed point integer.
This technique compresses the model and it affects the performance of the
model.

You know the idea is to find a balance between speed, accuracy and model
size so that the model will fit into the flash and ram respectively. This
is very difficult to achieve.

I cannot remember the precise amount of drop in accuracy, but yes, the
accuracy dropped when I ran it on the board.

You can look into STM32 CubeAI software expansion (it supports Keras,
TFlowLite, Caffe, ONNX etc). Alot of improvement has been made ever since
and you can achieve a better result using this.

I hope it helps.

Cheers.

joshbindels, 2020-11-08:
This looks very promising!

Thank you, I really appreciate it!

youngolutosin, 2020-11-08:
You are welcome.

beyonson, 2021-07-20:
Hello @marcin-ch 

I know this is a very old project for you, but I am trying to figure out this camera for another project and I am having trouble with the stm32 DCMI. I was wondering if you could upload the generated bin file you get when running the lines:

cd cmsisnn_demo/
mbed compile -m DISCO_F746NG -t GCC_ARM --source . --source ../ML-examples/cmsisnn-cifar10/camera_demo/camera_app/ 

Although my end goal is ML, I really need to figure out how to use the camera first. I would really appreciate any advice or help, thanks!!

NarendraKumarMadireddy, 2023-08-14:
> Hi @youngolutosin ,
> I followed this guide [Image recognition on Arm Cortex-M with CMSIS-NN guide](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/before-you-begin) as mentioned in my first post here. You can download this guide as PDF file (please see "Download PDF" option).
> 

Can you share your project report, I am working image recognition using cortex processor with cmsis library, can you help me in completing the project, I don't have much knowledge on this topic.

marcin-ch, 2023-08-14:
@NarendraKumarMadireddy seems this is the one
[Image recognition on Arm Cortex-M with CMSIS-NN.pdf](https://github.com/ARM-software/ML-examples/files/12339969/Image.recognition.on.Arm.Cortex-M.with.CMSIS-NN.pdf)
and Youtube video as well which might be helpful
[Image recognition on Arm Cortex-M with CMSIS-NN in 5 steps](https://www.youtube.com/watch?v=EkYp0glSenE)

youngolutosin, 2023-08-15:
It’s been awhile. I really do not have access to this project anymore.

NarendraKumarMadireddy, 2023-08-15:
> Hello All,
>
> I am following this [Image recognition on Arm Cortex-M with CMSIS-NN [...]

Can you share your project report, I am working on this image recognition using cortex processor with cmsis library, can you help me in completing the project, I don't have much knowledge on this topic.

Can you share the correct detailed step by step process.

NarendraKumarMadireddy, 2023-08-15:
> Hi @youngolutosin @marcin-ch @shell0108 ,
>
> I'm currently working on this project for my thesis as well.
>
> Did you notice a difference when running the model on the board or on a laptop?
> When I run [...]

Can you share your project report, I am working image recognition using cortex processor with cmsis library, can you help me in completing the project, I don't have much knowledge on this topic.

Can you share me the step by step procedure 🥺 please

youngolutosin, 2023-08-15:
I don’t have access to the project anymore. Please sto spamming.

NarendraKumarMadireddy, 2023-08-15:
> @NarendraKumarMadireddy seems this is the one
> [Image recognition on Arm Cortex-M with [...]

Thankyou but can you share me some report or etc , I am new to Linux and caffe

NarendraKumarMadireddy, 2023-08-15:
> I don’t have access to the project anymore. Please sto spamming.
> 
> 
> > Hi @youngolutosin <https://github.com/youngolutosin> @marcin-ch
> > <https://github.com/marcin-ch> @shell0108 <https://github.com/shell0108> ,
> >
> > I'm currently working on this project for my thesis as well.
> >
> > Did you notice a difference when running the model on the board or on a
> > laptop?
> > When I run accuracy tests using caffe or as a python script on my host
> > machine I'm getting a decent accuracy. But, when I run a similar accuracy
> > test on the board the accuracy is quite low, ~30%.
> >
> > Can you share your project report, I am working image recognition using
> > cortex processor with cmsis library, can you help me in completing the
> > project, I don't have much knowledge on this topic.
> >
> > Can you share me the step by step procedure 🥺 please
> >
> >
> 
> ​
> *Regards,*
> 
> *Olutosin​ ​Ademola*
> 
> *+37253564286,*
> 
> *DevOps Engineer | ML Engineer*
> 
> *Tallinn, Estonia.*

Sorry , thankyou

NarendraKumarMadireddy, 2023-08-17:
Last question, do anyone have the windows version tutorial/PDFs/report of this project?