Arm-Examples/ML-examples #22: ARMNN mobilenet quant example has undefined functions

idenc, 2019-08-08:
I am trying to run mobilenetv1_quant_tflite.cpp but in the section where the input image is loaded and preprocessed, there seems to be some undefined functions/types. On line 119 NormalizationParameters cannot be found and neither can PrepareImageTensor. I have all the includes and libraries set. Am I missing something?

lsc1234582, 2019-08-09:
Hi Iden! Thanks for raising this problem. The example code relies on ArmNN 19.08 release, which should be available on the 30th August (I will make that explicit on README). Alternatively, you can pull the developement repo from here: https://review.mlplatform.org/#/admin/projects/ml/armnn. (You can refer to the contribution page here: https://mlplatform.org/contributing/)

xmuspeech, 2019-08-12:
hello.I have a problem。mobilenetv1_quant_tflite.cpp:11:57: fatal error: ImageTensorGenerator/ImageTensorGenerator.hpp: No such file or directory

idenc, 2019-08-12:
I built the development repo but now I get an undefined reference to InferenceTestImage::Resize in PrepareImageTensor<float>.

lsc1234582, 2019-08-13:
Hi @xmuspeech. You need the 19.08 ArmNN release, which will be available on the 30th August. Alternatively, you can grab the dev repo at https://review.mlplatform.org/#/admin/projects/ml/armnn. 

@idenc. Did you build ArmNN with '-DBUILD_TESTS' option? The InferenceTestImage needs that. But that's a good catch, I'll make that explicit in the README as well.

idenc, 2019-08-14:
Thanks for the help but I am still unable to get the program to run correctly. I built with the tests and am able to compile the program but when I run it I get this output:
```LD_LIBRARY_PATH=$LD_LIBRARY_PATH:. ./mobilenetv1_quant_tflite -m "/home/pi/Downloads/mobilenet_quant/mobilenet_v1_0.25_224_quant/mobilenet_v1_0.25_224_quant.tflite" -d panda.jpg -p labels.txt -c CpuAcc CpuRef
ArmNN v20190500

WARNING: The given backend path is empty
Running network...
terminate called after throwing an instance of 'armnn::Exception'
  what():  Output tensor has incorrect size
Aborted
```

nextcam, 2019-08-15:
Check your labels file. The sample code uses the number of lines at the label file to allocate the correct output vector size.

`   // Output tensor size is equal to the number of model output labels
    const unsigned int outputNumElements = modelOutputLabels.size();
    std::vector<TContainer> outputDataContainers = { std::vector<uint8_t>(outputNumElements) };`

The labels.txt should be something like:

`background
tench
goldfish
great white shark
tiger shark
hammerhead
electric ray
...`

idenc, 2019-08-15:
Thanks! My labels.txt seemed to be missing a line somehow. Works now.

closed 2019-08-15 by idenc