Arm-Examples/ML-examples #38: Problem with numpy in pinet.py

Sqeak, 2020-04-13:
Hello! 
I've just been following the Yeah-World tutorial and have done all the things it said, except I changed the code in record.py so it can use a standard USB webcam:
 '  cap = cv2.VideoCapture(0)
    frames = []
    started = time()
    while time() - started < seconds:
        frame = cap.read()
        frames.append(frame)
'
The only thing I changed was the loop where it records frame by frame, I removed the camera dependencies from everything because it pointed towards Picamera, because I don't have a Picamera currently. After that I tested record.py and it worked! When I tried to train a model based on the standard yeah, sitting, and random, I got the error from pinet.py:
`
2020-04-13 15:37:00.067761: E tensorflow/core/platform/hadoop/hadoop_file_system.cc:132] HadoopFileSystem load error: libhdfs.so: cannot open shared object file: No such file or directory
Loading tensorflow feature extractor...
WARNING:tensorflow:From /home/pi/ML-examples/yeah-world/pinet.py:28: The name tf.gfile.GFile is deprecated. Please use tf.io.gfile.GFile instead.

WARNING:tensorflow:From /home/pi/ML-examples/yeah-world/pinet.py:29: The name tf.GraphDef is deprecated. Please use tf.compat.v1.GraphDef instead.

WARNING:tensorflow:From /home/pi/ML-examples/yeah-world/pinet.py:37: The name tf.Session is deprecated. Please use tf.compat.v1.Session instead.

Loading example/yeahTraceback (most recent call last):
  File "train.py", line 101, in <module>
    main()
  File "train.py", line 55, in main
    features = [feature_extractor.features(f) for f in x]
  File "train.py", line 55, in <listcomp>
    features = [feature_extractor.features(f) for f in x]
  File "/home/pi/ML-examples/yeah-world/pinet.py", line 45, in features
    preprocessed = ((np.array(image, dtype=np.float32) / 255.) - 0.5) * 2.
ValueError: setting an array element with a sequence.
`
I looked up the error and it said that it was trying to force and uneven multidimensional array into something? I don't know if it's messed up because my webcam may be giving it the wrong format of data, or if I don't have the right version of NumPy installed. Please forgive me if I did something wrong, this is my first time posting a question/issue

Sqeak, 2020-04-13:
And is there a way i can remove the TensorFlow errors, I've tried other versions but they don't work

mark-arm, 2020-04-22:
Easy things first: the TensorFlow warnings are just future compatibility warnings and can be ignored. If you want them to go away you can modify the lines of code referred to as described, e.g. change tf.Session to tf.compat.v1.Session and so on.

The error in preprocessing the image suggests the changes you made to use a USB webcam are not producing the same kind of output as is expected. I suggest printing out one of the frames and/or its shape and checking that it really is an array of width x height x depth. The fact that record.py runs without error is misleading - it just saves the data to a file, it doesn't care whether it is in the right format or not.

Sqeak, 2020-04-22:
Thank you! My picamera eventually arrived in the mail and everything worked flawlessly!

closed 2020-04-22 by Sqeak

ZissisT, 2020-07-23:
I had the same and just disabled V2 behavior in pinet.py:

```
import tensorflow.compat.v1 as tf
tf.disable_v2_behavior()
```

For webcam to work I had to change code like this in record.py:

```
#from camera import Camera
import cv2

def main():
.
.
    camera = cv2.VideoCapture(0)
    camera.set(cv2.CAP_PROP_FPS, 30)
    record(camera, name, seconds)
.
.
.
.
def record(camera, filename, seconds):
        ret, frame = camera.read()
        if ret == False:
            print("Failed to retrieve frame")
            break 
        frame = cv2.flip(frame, 1)
        frame = cv2.rotate(frame, cv2.ROTATE_90_COUNTERCLOCKWISE)
        frame = cv2.resize(frame, (128, 128), interpolation = cv2.INTER_AREA)
        frames.append(frame)
.
.
    dump(frames, open(filename, 'wb'), protocol=2)
    camera.release() 
    cv2.destroyAllWindows()
    print('done.')
.
.
```

throws some deprecation warnings, but works for me. Hope it helps others who want to use webcam with newer tf installed.