Arm-Examples/ML-examples #57: Error while running test_tflite.py

mehreenjabeen, 2021-07-01:
Hi,
while running test_tflite.py, I got the below error 
_"ValueError: Cannot set tensor: Dimension mismatch. Got 3920 but expected 490 for dimension 1 of input 37."_

Command i ran:
 _python test_tflite.py --tflite_path ../Pretrained_models/DS_CNN/DS_CNN_S/ds_cnn_s_quantized.tflite_

Burton2000, 2021-07-01:
Hi there,

You need to set the parameters to match the model being used as DNN and DS_CNN have different input features. The default values are for the DNN model but you are testing the ds_cnn_s model. The following should hopefully work for you:

`python test_tflite.py --tflite_path ../Pretrained_models/DS_CNN/DS_CNN_S/ds_cnn_s_qantized.tflite --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20`

mehreenjabeen, 2021-07-02:
It worked. Thanks Richard.

closed 2021-07-02 by mehreenjabeen