Arm-Examples/ML-examples #59: make -f tensorflow/lite/micro/tools/make/Makefile TAGS="DS_CNN Simple_KWS_Test" generate_kws_cortex_m_mbed_project

alcoal1, 2021-07-22:
I am following the tutorial here:[tflu-kws-cortex-m](https://github.com/ARM-software/ML-examples/tree/master/tflu-kws-cortex-m/kws_cortex_m) 
$ make -f tensorflow/lite/micro/tools/make/Makefile TAGS="DS_CNN Simple_KWS_Test" generate_kws_cortex_m_mbed_project
This produces the following error:
make: *** no rules to make target“tensorflow/lite/micro/tools/make/gen/linux_x86_64/prj/kws_cortex_m/mbed/third_party/gemmlowp/fixedpoint/fixedpoint.h”， due to “generate_kws_cortex_m_mbed_project”  stop

How to resolved this error?

Burton2000, 2021-07-26:
Hmm could be related to TensorFlow version used. Before you call make could try running this command to check out a specific version of TensorFlow (they deprecated some things in the latest versions).

```git checkout 72c19e8880```

It might also make things easier to re-clone the tensorflow repository fresh and copy in the kws_cortex_m sub folder again.

We have an open PR for this repository that will be merged very soon hopefully https://github.com/ARM-software/ML-examples/pull/60 that includes some updates including this in the README.

Let me know if this helps.

alcoal1, 2021-07-27:
> Hmm could be related to TensorFlow version used. Before you call make could try running this command to check out a specific version of TensorFlow (they deprecated some things in the latest [...]

Thanks for response.

But I did it according to your advice and still encountered this error again.

Burton2000, 2021-07-27:
So just to confirm then, you have done the following:

```cd tflu-kws-cortex-m/
git clone https://github.com/tensorflow/tensorflow.git
cp -r ./kws_cortex_m tensorflow/tensorflow/lite/micro/examples/
cd tensorflow/
git checkout 72c19e8880
make -f tensorflow/lite/micro/tools/make/Makefile TAGS="DS_CNN Simple_KWS_Test" generate_kws_cortex_m_mbed_project```

And you get the error?

alcoal1, 2021-07-28:
@Burton2000 Thank you,I fixed this error.
It's my network fault.
I will close this issue.

But when i use command:
`mbed compile -m K64F -t GCC_ARM`

I get a new error, so i take a new [issue](https://github.com/ARM-software/ML-examples/issues/61) could you please help me?

closed 2021-07-28 by alcoal1