Arm-Examples/ML-examples #61: Error in compile tflu-kws-cortex-m/kws_cortex_m

alcoal1, 2021-07-28:
I am following the tutorial with these command:
`make -f tensorflow/lite/micro/tools/make/Makefile TAGS="DS_CNN Simple_KWS_Test" generate_kws_cortex_m_mbed_project`
`cd tensorflow/lite/micro/tools/make/gen/linux_x86_64/prj/kws_cortex_m/mbed/`

change the mbed-os.lib version : 
`https://github.com/ARMmbed/mbed-os/#60cbab381dd2d5d860407b1b789741275012a075`

`mbed config root .`
`mbed deploy`
`mbed compile -m K64F -t GCC_ARM`

and then t get an error:

`Compile [ 99.2%]: arm_rfft_fast_init_f32.c
Compile [ 99.6%]: arm_rfft_init_f32.c
Compile [100.0%]: schema_utils.cc
Link: mbed
/home/fjw/download/gcc-arm-none-eabi/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: BUILD/K64F/GCC_ARM/mbed-os/platform/source/mbed_sdk_boot.o: in function `__wrap_main':
/home/fjw/ML-examples/tflu-kws-cortex-m/tensorflow/tensorflow/lite/micro/tools/make/gen/linux_x86_64/prj/kws_cortex_m/mbed/./mbed-os/platform/source/mbed_sdk_boot.c:177: undefined reference to `main'
collect2: error: ld returned 1 exit status
[ERROR] /home/fjw/download/gcc-arm-none-eabi/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: BUILD/K64F/GCC_ARM/mbed-os/platform/source/mbed_sdk_boot.o: in function `__wrap_main':
/home/fjw/ML-examples/tflu-kws-cortex-m/tensorflow/tensorflow/lite/micro/tools/make/gen/linux_x86_64/prj/kws_cortex_m/mbed/./mbed-os/platform/source/mbed_sdk_boot.c:177: undefined reference to `main'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "/home/fjw/anaconda3/bin/python" returned error.
       Code: 1
       Path: "/home/fjw/ML-examples/tflu-kws-cortex-m/tensorflow/tensorflow/lite/micro/tools/make/gen/linux_x86_64/prj/kws_cortex_m/mbed"
       Command: "/home/fjw/anaconda3/bin/python -u /home/fjw/ML-examples/tflu-kws-cortex-m/tensorflow/tensorflow/lite/micro/tools/make/gen/linux_x86_64/prj/kws_cortex_m/mbed/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
`
How can I solve this problem?

liabar01, 2021-07-29:
Can you try a fresh clone/install using regular Python and pip environments in place of Conda? 

It is also possible you are missing some board specific drivers/utilities; these are typically added via mbed add:

`mbed add https://os.mbed.com/teams/ST/code/BSP_DISCO_F746NG/`

What is the full name of your board? I do not see a counterpart for K64F but let me know if using pip in place of conda environments does not resolve the issue and I can investigate further.

What command do you run to change the mbed OS version?

alcoal1, 2021-08-02:
@liabar01 Thanks for you advice.
I reconfigured the environment using Conda and rerun succeed.

closed 2021-08-02 by alcoal1

Omer-sync, 2022-10-26:
> @liabar01 Thanks for you advice. I reconfigured the environment using Conda and rerun succeed.

Hello,
How did you exactly solve this particular problem? I am also stuck with this now and I can not find a solution. 
Thanks in advance.