Arm-Examples/ML-examples #62: couldn't retrieve the checkpoints

Aishaj, 2021-07-28:
Hello, 

I'm trying to run the following line
`python test.py --model_architecture dnn --model_size_info 128 128 128 --checkpoint /tflu-kws-cortex-m/Pretrained_models/DNN/DNN_S/ckpt/checkpoint
`

Howevre I'm getting the following error 

`Traceback (most recent call last):
  File "test.py", line 182, in <module>
    test()
  File "test.py", line 44, in test
    model.load_weights(FLAGS.checkpoint).expect_partial()
  File "/home/pi/miniconda3/lib/python3.6/site-packages/tensorflow/python/keras/engine/training.py", line 162, in load_weights
    return super(Model, self).load_weights(filepath, by_name)
  File "/home/pi/miniconda3/lib/python3.6/site-packages/tensorflow/python/keras/engine/network.py", line 1384, in load_weights
    pywrap_tensorflow.NewCheckpointReader(filepath)
  File "/home/pi/miniconda3/lib/python3.6/site-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 636, in NewCheckpointReader
    return CheckpointReader(compat.as_bytes(filepattern))
  File "/home/pi/miniconda3/lib/python3.6/site-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 648, in __init__
    this = _pywrap_tensorflow_internal.new_CheckpointReader(filename)
tensorflow.python.framework.errors_impl.InvalidArgumentError: Unsuccessful TensorSliceReader constructor: Failed to get matching files on /tflu-kws-cortex-m/Pretrained_models/DNN/DNN_S/ckpt/checkpoint: Not found: /tflu-kws-cortex-m/Pretrained_models/DNN/DNN_S/ckpt; No such file or directory
`

Any anyone had the same issue and managed to fix it? 

Thank you

Burton2000, 2021-07-28:
Your path is incorrect, you have supplied an absolute path so it is looking in your whole machine for ```/tflu-kws-cortex-m/Pretrained_models/DNN/DNN_S/ckpt```
 you are probably wanting to use a relative path like ```../Pretrained_model/Dnn/DNN_S/ckpt/dnn_0.84_ckpt``` instead.

Also note that when using the checkpoint files they correspond to a model in train_commands.txt so you need to make sure parameters you supply to test.py like model_size_info, window_size_ms, winow_stride_ms match what was used in training these checkpoints. You can just take what is used in train_commands.txt for this bit.

Aishaj, 2021-07-29:
Thank you Richard @Burton2000 . I edited the path and it worked.  
I did use the parameters in train_commands.txt but still there is a problem. When running this 
`python test.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --checkpoint ../Pretrained_models/DNN/DNN_S/ckpt/dnn_0.84_ckp`

I'm getting , 
`Running testing on validation set...
<DatasetV1Adapter shapes: ((?, 3920), (?,)), types: (tf.float32, tf.int32)>`

Thank you

Burton2000, 2021-07-29:
The parameters should match the checkpoint you are trying to restore. Those parameters are for the DS-CNN S model but you are trying to restore the DNN-S checkpoint. You would want something like this and it should work:

```
python test.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --dct_coefficient_count 10 --window_size_ms 40 --window_stride_ms 20 --checkpoint ../Pretrained_models/DS_CNN/DS_CNN_S/ckpt/ds_cnn_0.94_ckpt
```

Aishaj, 2021-07-30:
Hi again @Burton2000  

Unfortunately still not working. I was working on a raspberry pi and now I changed to windows but still not working. This is the error I'm getting. 

`C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\numpy\_distributor_init.py:32: UserWarning: loaded more than 1 DLL from .libs:
C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\numpy\.libs\libopenblas.TXA6YQSD3GCQQC22GEQ54J2UDCXDXHWN.gfortran-win_amd64.dll
C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\numpy\.libs\libopenblas.WCDJNK7YVMPZQ2ME2ZZHJJRJ3JIKNDB7.gfortran-win_amd64.dll
  stacklevel=1)
2021-07-30 21:40:08.964359: W tensorflow/stream_executor/platform/default/dso_loader.cc:60] Could not load dynamic library 'cudart64_110.dll'; dlerror: cudart64_110.dll not found
2021-07-30 21:40:08.964751: I tensorflow/stream_executor/cuda/cudart_stub.cc:29] Ignore above cudart dlerror if you do not have a GPU set up on your machine.
2021-07-30 21:40:19.711095: I tensorflow/compiler/jit/xla_cpu_device.cc:41] Not creating XLA devices, tf_xla_enable_xla_devices not set
2021-07-30 21:40:19.713635: W tensorflow/stream_executor/platform/default/dso_loader.cc:60] Could not load dynamic library 'nvcuda.dll'; dlerror: nvcuda.dll not found
2021-07-30 21:40:19.734986: W tensorflow/stream_executor/cuda/cuda_driver.cc:326] failed call to cuInit: UNKNOWN ERROR (303)
2021-07-30 21:40:19.800311: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:169] retrieving CUDA diagnostic information for host: DESKTOP-S9FFEFJ
2021-07-30 21:40:19.800638: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:176] hostname: DESKTOP-S9FFEFJ
2021-07-30 21:40:19.827560: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2021-07-30 21:40:19.830795: I tensorflow/compiler/jit/xla_gpu_device.cc:99] Not creating XLA devices, tf_xla_enable_xla_devices not set
Untarring speech_commands_v0.02.tar.gz...
Running testing on validation set...
Traceback (most recent call last):
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\util\dispatch.py", line 201, in wrapper
    return target(*args, **kwargs)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\ops\math_ops.py", line 1786, in tensor_not_equals
    return gen_math_ops.not_equal(self, other, incompatible_shape_error=False)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\ops\gen_math_ops.py", line 6401, in not_equal
    ctx=_ctx)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\ops\gen_math_ops.py", line 6429, in not_equal_eager_fallback
    _attr_T, _inputs_T = _execute.args_to_matching_eager([x, y], ctx, [])
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\eager\execute.py", line 274, in args_to_matching_eager
    t, dtype, preferred_dtype=default_dtype, ctx=ctx)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\profiler\trace.py", line 163, in wrapped
    return func(*args, **kwargs)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\ops.py", line 1540, in convert_to_tensor
    ret = conversion_func(value, dtype=dtype, name=name, as_ref=as_ref)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\constant_op.py", line 339, in _constant_tensor_conversion_function
    return constant(v, dtype=dtype, name=name)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\constant_op.py", line 265, in constant
    allow_broadcast=True)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\constant_op.py", line 276, in _constant_impl
    return _constant_eager_impl(ctx, value, dtype, shape, verify_shape)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\constant_op.py", line 301, in _constant_eager_impl
    t = convert_to_eager_tensor(value, ctx, dtype)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\constant_op.py", line 98, in convert_to_eager_tensor
    return ops.EagerTensor(value, ctx.device_name, dtype)
ValueError: TypeError: object of type 'RaggedTensor' has no len()

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 182, in <module>
    test()
  File "test.py", line 48, in test
    val_data = audio_processor.get_data(audio_processor.Modes.VALIDATION).batch(FLAGS.batch_size)
  File "C:\Users\XXX\Dropbox\Documents\PhD\Coding\KWS\tflu-kws-cortex-m\Training\data.py", line 190, in get_data
    use_background = (self.background_data != []) and (mode == AudioProcessor.Modes.TRAINING)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\util\dispatch.py", line 205, in wrapper
    result = dispatch(wrapper, args, kwargs)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\util\dispatch.py", line 118, in dispatch
    result = dispatcher.handle(args, kwargs)
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\ops\ragged\ragged_dispatch.py", line 219, in handle
[... cut ...]
    return _add_should_use_warning(fn(*args, **kwargs),
  File "C:\Users\XXX\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\ops\control_flow_ops.py", line 158, in Assert
    (condition, "\n".join(data_str)))
tensorflow.python.framework.errors_impl.InvalidArgumentError: Expected 'tf.Tensor(False, shape=(), dtype=bool)' to be true. Summarized data: b'Unable to broadcast: dimension size mismatch in dimension'
1
b'lengths='
0
b'dim_size='
1522930, 988891, 980062, 960000, 978488, 960000`

closed 2021-08-05 by Aishaj