Arm-Examples/ML-examples #65: Freez.py for tensrflow2.5 versin

Aishaj, 2021-08-05:
Hi, 
I'm trying to edit the freez.py from the main project folder( which is Tensorflow 1.X). to make it suitable for thates project files( Tensorflow 2.5)  However, when running
`python freeze.py --model_architecture ds_cnn --model_size_info 5 64 10 4 2 2 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 64 3 3 1 1 --checkpoint ../Pretrained_models/DS_CNN/DS_CNN_S/ckpt/ds_cnn_0.94_ckpt --output_file ds_cnn.pb`

I'm getting the following error

2021-08-05 15:42:18.490308: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2021-08-05 15:42:18.491455: I tensorflow/core/common_runtime/process_util.cc:146] Creating new thread pool with default inter op setting: 2. Tune using inter_op_parallelism_threads for best performance.
WARNING:tensorflow:From C:\Users\ash_j\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\keras\layers\normalization.py:534: _colocate_with (from tensorflow.python.framework.ops) is deprecated and will be removed in a future version.
Instructions for updating:
Colocations handled automatically by placer.
W0805 15:42:18.584352 17588 deprecation.py:336] From C:\Users\ash_j\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\keras\layers\normalization.py:534: _colocate_with (from tensorflow.python.framework.ops) is deprecated and will be removed in a future version.
Instructions for updating:
Colocations handled automatically by placer.
Traceback (most recent call last):
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\tensor_util.py", line 549, in make_tensor_proto
    str_values = [compat.as_bytes(x) for x in proto_values]
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\tensor_util.py", line 549, in <listcomp>
    str_values = [compat.as_bytes(x) for x in proto_values]
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\util\compat.py", line 87, in as_bytes
    (bytes_or_text,))
TypeError: Expected binary or unicode string, got <tensorflow.python.keras.engine.functional.Functional object at 0x000002E6F9A16F08>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "freeze.py", line 248, in <module>
    tf.compat.v1.app.run(main=main, argv=[sys.argv[0]] + unparsed)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\platform\app.py", line 40, in run
    _run(main=main, argv=argv, flags_parser=_parse_flags_tolerate_undef)
  File "C:\Users\x\xanaconda3\envs\newenvt\lib\site-packages\absl\app.py", line 312, in run
    _run_main(main, args)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\absl\app.py", line 258, in _run_main
    sys.exit(main(argv))
  File "freeze.py", line 176, in main
    FLAGS.model_size_info)
  File "freeze.py", line 115, in create_inference_graph
    tf.nn.softmax(logits, name='labels_softmax')
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\util\dispatch.py", line 206, in wrapper
    return target(*args, **kwargs)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\ops\nn_ops.py", line 3701, in softmax_v2
    return _wrap_2d_function(logits, gen_nn_ops.softmax, axis, name)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\ops\nn_ops.py", line 3613, in _wrap_2d_function
    inputs = ops.convert_to_tensor(inputs)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\profiler\trace.py", line 163, in wrapped
    return func(*args, **kwargs)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\ops.py", line 1566, in convert_to_tensor
    ret = conversion_func(value, dtype=dtype, name=name, as_ref=as_ref)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\constant_op.py", line 339, in _constant_tensor_conversion_function
    return constant(v, dtype=dtype, name=name)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\constant_op.py", line 265, in constant
    allow_broadcast=True)
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\constant_op.py", line 283, in _constant_impl
    allow_broadcast=allow_broadcast))
  File "C:\Users\x\anaconda3\envs\newenvt\lib\site-packages\tensorflow\python\framework\tensor_util.py", line 553, in make_tensor_proto
    "supported type." % (type(values), values))
TypeError: Failed to convert object of type <class 'tensorflow.python.keras.engine.functional.Functional'> to Tensor. Contents: <tensorflow.python.keras.engine.functional.Functional object at 0x000002E6F9A16F08>. Consider casting elements to a supported type.

Would appreciate any help regarding this.
Thanks

Burton2000, 2021-08-05:
Freeze.py in the original repository was for generating a frozen .pb model file for inference. This model format is exclusive to TensorFlow 1 and is not supported in TensorFlow 2.

In this repository you can instead generate a TFLite file using ```convert.py``` that will be suitable for deployment. You can pass in ```--no_quantize``` to generate a floating point file if you wish so that it matches the .pb model.

Aishaj, 2021-08-05:
The reason of me trying to create the freez.py is because I want to use label_wav.py. So I want to send a sound file to the model. However, label_wav.py requires a .pb file

Burton2000, 2021-08-31:
This merge request https://github.com/ARM-software/ML-examples/pull/72 has a new label_wav.py file that should work with tflite files, give it a try and see if it works for you.