Arm-Examples/ML-examples #104: Error in compile tflu-kws-cortex-m/kws_cortex_m

charlescho64, 2022-11-24:
Hello...

I am following with guide in this site ii order to deploy AI model to STM32F4. But I have the problem in compile. 
This issue is the same to other issue(Error in compile tflu-kws-cortex-m/kws_cortex_m #61) but the solution is not open. 

I am in problem with compile under cross compile 9.3.1 and python 3.9.15 on osx mac.

-------------
Compile [ 99.7%]: test_helpers.cc
Compile [100.0%]: schema_utils.cc
Link: mbed
/Applications/ARM/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: BUILD/DISCO_H747I/GCC_ARM/mbed-os/platform/source/mbed_sdk_boot.o: in function `__wrap_main':
/Users/charles/Documents/DocPrj/NotYet/ML-examples/tflu-kws-cortex-m/tensorflow/tensorflow/lite/micro/tools/make/gen/osx_x86_64/prj/kws_cortex_m/mbed/./mbed-os/platform/source/mbed_sdk_boot.c:177: undefined reference to `main'
collect2: error: ld returned 1 exit status
[ERROR] /Applications/ARM/bin/../lib/gcc/arm-none-eabi/9.3.1/../../../../arm-none-eabi/bin/ld: BUILD/DISCO_H747I/GCC_ARM/mbed-os/platform/source/mbed_sdk_boot.o: in function `__wrap_main':
/Users/charles/Documents/DocPrj/NotYet/ML-examples/tflu-kws-cortex-m/tensorflow/tensorflow/lite/micro/tools/make/gen/osx_x86_64/prj/kws_cortex_m/mbed/./mbed-os/platform/source/mbed_sdk_boot.c:177: undefined reference to `main'
collect2: error: ld returned 1 exit status

[mbed] ERROR: "/Users/charles/Documents/DocPrj/NotYet/ML-examples/tflu-kws-cortex-m/.mbedenv/bin/python" returned error.
       Code: 1
       Path: "/Users/charles/Documents/DocPrj/NotYet/ML-examples/tflu-kws-cortex-m/tensorflow/tensorflow/lite/micro/tools/make/gen/osx_x86_64/prj/kws_cortex_m/mbed"
       Command: "/Users/charles/Documents/DocPrj/NotYet/ML-examples/tflu-kws-cortex-m/.mbedenv/bin/python -u /Users/charles/Documents/DocPrj/NotYet/ML-examples/tflu-kws-cortex-m/tensorflow/tensorflow/lite/micro/tools/make/gen/osx_x86_64/prj/kws_cortex_m/mbed/mbed-os/tools/make.py -t GCC_ARM -m DISCO_H747I --source . --build ./BUILD/DISCO_H747I/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
-------------------

Burton2000, 2022-12-05:
From the issue you linked to the error seems to have been caused by the Python environment. Are you able to try creating a fresh Python environment, installing mbed-cli and trying building again?

mesut92, 2022-12-28:
I got same error. What is your Python  version. Does it need a specific Python Version. Or, is there any requirements file for conda enviroment?